emoncms/app #3: app_config mysql table is missing

yekorzh, 2015-07-23:
Hi,

I've installed apps on my own emoncms instance and noticed that app configs where not saved. After some digging in the app sources I found that the config gets written into app_config MySQL table, which was missing on my emoncms. I also couldn't find any place in the app code where emoncms would try creating that table in the database.

I suggest that AppConfig_model.php should create that table automatically if it doesn't exist.

TrystanLea, 2015-07-24:
Hello yekorzh, the code for creating the table is in the core, you need to run the database update tool from the Admin tab in emoncms and then it should work ok.

yekorzh, 2015-07-24:
Trystan,

Thank you, I guess it is better to mention somewhere in the doco then? I don't remember if I saw it anywhere...

Yegor

glynhudson, 2019-01-24:
The database update will run automatically as part of an emonPi / emonBase update. If updating manually running a manual DB update as described above can be required.

closed 2019-01-24 by glynhudson