emoncms/app #13: My Electric app not updating

AndyRap, 2016-03-26:
Hi,

I have a problem whereby the app does not update until I refresh the screen.

Looking at the developer console I see the following error -

[image]

which seems to be tied to the data refresh period -

[image]

Any advice on how to correct the issue would be welcome.

Thanks in advance
Andy Hodges

AndyRap, 2016-03-26:
Fixed,

I don't know JavaScript, but......

Removing the [0] from line 516 -

var next = valid[valid.length-1][0] + (interval*1000);

so as to leave the line as -

var next = valid[valid.length-1] + (interval*1000);

Has fixed the issue.

Kind Regards
Andy Hodges

glynhudson, 2016-03-29:
Hi Andy, Thanks for reporting issue. I have been unable to re-create. 

What version of Emoncms are you running? 
What browser are you using? 
If using an emonBase/emonPi have you updated to latest version using emonPi update? 

Please could you provide some more info regarding the issue:

Does MyElectric no update at all?
 Or does it stop updating after a period of time?
When does the error in the console appear?

AndyRap, 2016-03-29:
Hi Glyn,

What version of Emoncms are you running? 
Emoncms 9.4

What browser are you using? 
Edge and Chrome, the issue existed in both.

If using an emonBase/emonPi have you updated to latest version using emonPi update? 
N/A, vanilla Pi build as per method below.

Does MyElectric no update at all?
MyElectric would update if I clicked the refresh button or if I chose a different time view for the bar graph but not if I just let it sit there.

Or does it stop updating after a period of time?
As soon as the page is loaded the value would not change without interaction as per the above answer.

When does the error in the console appear? 
The console error appeared as soon as the MyElectric page was loaded.

_METHOD_
I am running on a Raspberry Pi 2.  All code was fresh from github and was installed on a fresh and updated Raspbian Jessie.

1.  Followed EmonCMS instructions for installing to a Pi, skipped the EmonHub installation.
2. Created new user, inputs, feeds etc.
3.  Installed the dashboard module using cd /var/www/emoncms/Modules && git clone https://github.com/emoncms/dashboard.git
4. Ran database update twice, as required.
5. Installed the app module using cd /var/www/emoncms/Modules && git clone https://github.com/emoncms/app.git
6. Ran the database update once, as required.
7. Set the feeds (Power and kWh) and the price per unit and saved.

That is what I did to get to the point that generated the error in the Microsoft Edge developer console and, as I know MS stuff can be fussy I also checked but didn't screenshot, the same error in Google Chrome developer console.

The error also existed on the WAMP/LAMP Windows 10 install I have setup to do some fiddling / dev work.  I will change the file back and see if I can reproduce the issue.

Right, just done that and it updates :8ball: 

The error only exists if the feeds that have been set have less than one days data in them, that is, the feed has not yet gone past midnight.

I have just created a temp power and kWh feed and assigned them to the MyElectric app and the error has returned.

Kind Regards
Andy Hodges

Kind Regards
Andy Hodges

glynhudson, 2016-03-30:
Thanks, fixed has been pushed :+1:

commit 09fd4f61 (2016-03-30): Fix #13 MyElectric does not update if  if the feeds that have been set have less than one days data in them (not past midnight)
closed 2016-03-30 by glynhudson