emoncms/app #15: Kwh Use displays zero until first 24hrs has past

glynhudson, 2016-08-14:
When first setting up a system if the feed has been running for less than 24hrs the kwh per day on the MyElectric app displays zero when in fact we have used 2.01Kwh (see below). Feed selecting in MyElectric app is correct

[image]

glynhudson, 2017-02-10:
https://community.openenergymonitor.org/t/ios-app-release/2134/16?u=glyn.hudson

gerg7, 2018-02-07:
Hi,
Seems this is still not corrected - how about following code change in myelectric.php :
line 584:     
    if (daily.length>0) {
        usetoday_kwh = daily[daily.length-1][1];
    } **else {
        usetoday_kwh = feeds[use_kwh].value*1.0;
    }**

PR #65 (2019-02-05): Fix not working when feed is <24 hours old
This fixes the long-standing bug where MyElectric and MyElectric2 don't
work properly if the feed you use for kwh values is <24 hours old. It
simply adds in the relevant data for today midnight and tomorrow
midnight to make the calculations work.

Fixes #15.
commit 55abdfef (2019-02-05): Fix not working when feed is <24 hours old

This fixes the long-standing bug where MyElectric and MyElectric2 don't
work properly if the feed you use for kwh values is <24 hours old. It
simply adds in the relevant data for today midnight and tomorrow
midnight to make the calculations work.

Fixes #15.
closed 2019-02-05 by TrystanLea