emoncms/app #21: Can't select Node:0:PAPP input feed

arthurlutz, 2016-10-17:
In the dropdown menu of "My Electric" I can't seem to select an input feed called by "Node:0:PAPP" all my "Node:0" feeds don't appear, the "Node 2" "Node 3" feeds appear so I guess there is a problem with the presence of the the ":" character (feed created on a older version of emoncms).

Is this a bug in app ? should I rename my Node:0 feeds and if so, how do I do it without loosing the data ?

arthurlutz, 2016-10-17:
Trying to set autoname in myelectric.js I get the following error on https://URL/emoncms/app/setconfig.json?data={"myelectric":{}} which generates a pop up, in firefox inspect I see : 

```
<br />
<b>Notice</b>:  Trying to get property of non-object in <b>/var/www/arthur/emoncms/Modules/app/AppConfig_model.php</b> on line <b>69</b><br />
<br />
<b>Fatal error</b>:  Call to a member function bind_param() on boolean in <b>/var/www/arthur/emoncms/Modules/app/AppConfig_model.php</b> on line <b>80</b><br />

```

glynhudson, 2016-10-18:
>s this a bug in app ? should I rename my Node:0 feeds and if so, how do I do it without loosing the data ?

This sounds like a bug in the app, thanks for reporting. You should be able to rename to remove the '.' by clicking on the pencil icon then editing the feed name on the Feeds page. The data and input processing for this feed will not be affected.

arthurlutz, 2016-10-24:
OK, renamed it but it hasn't changed the fact that I can't select it. 

My problem seems to be that my watt usage is in a PHPFIWA timeseries, not PHPINA. I'm off to look for tools to convert that.

glynhudson, 2016-10-24:
Please post on our community forum: https://community.openenergymonitor.org/

closed 2016-10-24 by glynhudson

pb66, 2016-10-24:
See the [Does emoncms expects data to be send reguarly? ]( https://community.openenergymonitor.org/t/does-emoncms-expects-data-to-be-send-reguarly/1884/7?u=pb66) forum thread

PHPfiwa support (in the menu list) was dropped by [removing the ",6" from the engine types](https://github.com/emoncms/app/commit/ada573b822a073a700f4643840ff0ec7ab7936a8#diff-253d0f62050ae92be3c76083d8acb087L5) in each of the app's .js files and can be put back for a quick solution until you change your feeds or permenantly.