emoncms/app #24: Add energy storage (e.g. display SOC)

martinjaeger, 2017-04-29:
As far as I can see, there is no possibility to show information about a battery system (for off-grid applications or for increased self-consumption).

I think it would be nice to show a graph of the State of Charge (SOC) in the "My Solar" app. Could be retrieved from a feed called "soc".

I'm currently only using emoncms.org, but will set up a local emoncms version soon and try to implement the energy storage integration.

Should I integrate it into the "My Solar" app or should I make another separate app, e.g. "My Off-Grid Solar"?

TrystanLea, 2017-04-29:
Hello Martin, good to hear from you! That would be a great addition. Im currently working through a large update to the app's module to make it easier to add new apps (hopefully), the new version is in the 'refactor' branch. Creating a new app is I think the best way to go about it initially and then depending on the level of changes we could then see whether to merge the two.

sysrqio, 2017-06-28:
@martin you could try https://github.com/emoncms/app/pull/30

closed 2026-10-06 by TrystanLea