emoncms/app #29: unitcost and currency not written to db

rexometer, 2017-06-27:
Hi,
the variables unitcost and currency are not written to the database. Trying to change this settings results in an error. Did look at the code, but didn't find the error till now ...
If I insert them manually into the database they are shown correctly, but modifying is still not possible.

rexometer, 2017-07-20:
Hi, I investigated on that a bit more: I was incorrect in my first post: unit cost works well.
And with currency it depends what symbol you type in: $ and £ are working **€ does not work**. Error message: `"Invalid config value"`
So it's obviously a problem with "special" characters. I just wonder why € is special and £ or $ not :)

greeebs, 2017-07-25:
While extended the existing time of use App, I found a need for config values that I could convert to multidimensional arrays. I found the list of characters that could be used for separators (and text strings!) to be much smaller than I expected.

Now I know why!

Is there any reason why ":" and "-" are also not in the list of accepted characters in the config fields?

greeebs, 2017-07-29:
@rexometer I added this to PR #35 which has now been merged.

closed 2017-07-30 by glynhudson