emoncms/app #33: Current power in kilowhatts

3raider, 2017-07-24:
Dear developers , is it possibe to add option (checkbox) to use current power not in whatts but in kW because f.e. power 1247 w looks not informative than 1,27 kW

greeebs, 2017-07-26:
How about use kW for values over 1kW and W for values under 1kW? 
I might add that to my next App and see what it looks like...

Which App are you using where you'd like to see this?

glynhudson, 2019-05-15:
Good idea, it would make sense to display kW for values over 999W. This is very much an issue for monitoring 100kW+ generation / consumption! Values like 100000W don't make sense!

greeebs, 2019-05-15:
I did actually try this within my time of use - flexible app and found (personally) the kW less informative than the W... Can I suggest that we switch to displaying kW for values above 9999 W?

glynhudson, 2019-05-15:
Ok sure, that makes sense. It did occur to me that it may be confusing to have both W and kW displayed on a App at the same time. Do you think we should auto change to kWh above a certain value or have a manual toggle to switch between W / kW?

greeebs, 2019-05-15:
It just occurred to me that an option to define the cut-over value gives the best of both worlds!
The option allows the user to specify a value at which the display unit changes from W to kW.

If you always want kW, just set the option to 0 and if you always want W set the option to 10000000 (or some other unwieldy number.

PR #109 (2019-09-10): Feature/issue 33 display power in kw
fix #33 
[image]

added checkbox to allow power to be displayed in kW

[image]
closed 2019-09-10 by TrystanLea