emoncms/app #41: OpenEVSE App

glynhudson, 2017-10-19:
After discussion with @chris1howell it would be great if the OpenEVSE app could display the following:

- Energy per session
- Cost per session
- Temperature graph
- Realtime state indicator (e.g. charging / EV connected / sleeping etc)
- Realtime current

OpenEVSE Wifi currently posts to following to Emoncms 

[image]

teslatradeup, 2019-05-19:
This would be awesome!! Was just thinking about this... I'd love an Android app that displays exactly the same information the LED display shows on the OpenEVSE... Perhaps as a banner at the top of my device. :)

glynhudson, 2019-05-19:
Are you aware the OpenEVSE already has a web interface? Here is a demo: https://openevse.openenergymonitor.org/

teslatradeup, 2019-05-19:
Yes I do. But it would be awesome to have an app that puts your current
state in the notification area on a mobile device with notification options.

Same info as what's on the OpenEVSE itself. :)

Is that information available somehow with credentials locally through an
API?

teslatradeup, 2019-05-20:
Would be awesome if we could see the info straight up like this:
[image]

glynhudson, 2019-05-20:
Agree, that would be *awesome*. We would need a native app for that. If you know anyone with Android development experience who's interested in the OpenEVSE project please put them in contact.