emoncms/app #46: Cost Comparison frozen and not able to click.

rcojoe, 2018-02-09:
Hey guys.  I'm trying to get the cost compare to work on the public site but it is frozen.  When I click on the wrench on the top right of the app,  nothing happens, like there is a javascript error.  I see these in the javascript log, so it looks like my settings aren't right but I can't get into the config area to fix.
ReferenceError: kwh is not defined

#32

rcojoe, 2018-02-09:
Running the function: feed.listbyidasync() does return a list of my feeds.

rcojoe, 2018-02-09:
Further debugging shows the first exception happening in 
bargraph_load(start, end)
where start = 1517633999999
end = 1518238799999

Its actually calling getdataDMY_time_of_use but the error says it is for  feed.getdataDMY

rcojoe, 2018-02-09:
This is related to the 24 vs 48 allowed datapoints in the split parameter.  If I only put 24 split data points in the split,  it returns data correctly.

TrystanLea, 2026-10-06:
Cost Comparison is now archived. Time of use costs are covered by Time of use flexible and the My Electric Flow tariff explorer. Closing.

closed 2026-10-06 by TrystanLea