emoncms/app #81: History button missing from MySolarDivert

glynhudson, 2019-04-24:
The MySolarDivert app should have a history button to view bar graph history in the same way as MySolarPV when testing the new sidebar version of the app module the history button is missing:

[image]

Old app with history buttons present:

[image]

glynhudson, 2019-04-24:
Form thread: https://community.openenergymonitor.org/t/mysolar-divert-app-setup/10741/7

emrysr, 2019-04-26:
@glynhudson fixed within the dark-sidebar & mobile resize branch in this pull request: https://github.com/emoncms/app/pull/79 

**CSS issue fixed**. the stylesheet needed updating to include the responsive css class names.

[image]

glynhudson, 2019-04-30:
Nice, I can confirm issue is fixed 👍 

[image]

closed 2019-04-30 by glynhudson