emoncms/app #84: SolarDivert: real-time solar gen figure incorrect

glynhudson, 2019-05-13:
The real-time solar PV generation figure seems to be incorrect. The graph is recording correctly e.g 

Solar is incorrectly displaying 1669W, the tooltip shows the actual generation of 1059W. My solar PV is 1200Wp therefore 1.6kW is impossible! 

[image]

Can you replicate this issue @mattjgalloway?

mattjgalloway, 2019-05-13:
Huh I’ve never seen this! Is it happening constantly? It’s weird how it’s exactly the same number as total use. And the numbers don’t add up at all. The export is calculated from the others, so I think the maths is correct but the display is wrong.

By any chance has the CSS ID of the solar been changed to be same ID as total use?

glynhudson, 2019-05-13:
Mmm I've just checked the feed ID's for solar and total use, and they are unique. The solar is reading the same as house. 

I think the tool tip is slightly behind the real-time in this screenshot :

[image]

[image]

At least if you can't replicate the issue it could be just something effecting my system.

mattjgalloway, 2019-05-13:
I don’t mean the IDs of the feeds, but of the HTML tag in the code.

Yeh I don’t have this problem but I also haven’t updated to the latest where there was the big refactor for the new menu etc.

mattjgalloway, 2019-05-13:
@glynhudson Yeh, look here:
https://github.com/emoncms/app/blob/master/apps/OpenEnergyMonitor/mysolarpvdivert/mysolarpvdivert.php#L166

That should be:
```
<span class="generationnow">
```

Looks like it was broken in this commit:
https://github.com/emoncms/app/commit/4f0d030467734ec929013be983c6840b72f48ce4

PR incoming...

mattjgalloway, 2019-05-13:
@glynhudson #85

glynhudson, 2019-05-15:
Fix by https://github.com/emoncms/app/pull/85

closed 2019-05-15 by glynhudson