emoncms/app #89: Adding tariffs to costcomparison app

delboy711, 2019-05-26:
The Cost Comparison app is nice, but it does not have many tariffs, and they are not up to date.
It would be nice if the user could add their own tariffs in a different file to rates.js that did not get overwritten with a new release of emoncms.

I added a line to costcomparison.php after line 18
`<script type="text/javascript" src="<?php echo $path; echo $appdir; ?>userrates.js?v=<?php echo $v; ?>"></script>`

Tariffs in the  file userrates.js get merged with the default tariffs

stuartpittaway, 2020-04-09:
@TrystanLea did you find a website with an api to get rates?

TrystanLea, 2026-10-06:
Cost Comparison is now archived. Time of use flexible has a tariff schedule builder for custom tariffs. Closing.

closed 2026-10-06 by TrystanLea