emoncms/app #100: Apps only list feeds with unique feed names

pb66, 2019-08-14:
see https://community.openenergymonitor.org/t/apps-feedlist-issues/11669?u=pb66

if two or more feeds have the same name (case sensitive) only the one with the lower feed id appears in the feedlist in the app config pages (tested on mysolarpv)

TrystanLea, 2019-08-14:
I think this may well have been an issue for a while, the apps module arranges the feeds in an associative array by name rather than by node:name combination, needs a good bit of work to change in the code but obviously worthwhile.

adminde, 2019-08-15:
This should be addressed in  #97.
I did not address the issues this raises regarding the auto selection though. And the visualization of the selected feed names looks still a bit strange, as the tag gets not yet displayed:

[image]

TrystanLea, 2020-01-21:
This is now fixed in latest master

closed 2020-01-21 by TrystanLea