emoncms/app #102: Inconsistently seeing this error

pb66, 2019-08-16:
[image]

It isn't occurring with any noticeable consistentcy and I'm not sure where it is pointing to. the app_view.php file doesn't have 735 lines! 

All latest versions of master branches as of today.

I'll post more if/when I can.

glynhudson, 2019-08-29:
I cannot seem to recreate that error on MySolar PV / MyEnergy app. However, I'm getting a similar error from the Cos Comparison app 

[image]

Looks like duplicate of https://github.com/emoncms/app/issues/62

glynhudson, 2019-08-29:
Another user has reported this error:

[image]

https://community.openenergymonitor.org/t/my-app/11800

pb66, 2019-08-30:
I've only seen this occur on my electric apps so a fix to the openevse app cannot possibly fix the issue I've experienced so  this should not be closed

glynhudson, 2019-09-04:
Ok, I've also just experienced this error when trying to create a new MyElectric 1 graph:

[image]

pb66, 2019-09-05:
Also see https://community.openenergymonitor.org/t/my-electric-app-error-on-session-expiry/11855?u=pb66

pb66, 2019-09-05:
> Ok, I've also just experienced this error when trying to create a new MyElectric 1 graph:
> 
> [image]

Although maybe related, the fault reported in your notification is slightly different and I experienced the issue on existing apps rather than when creating new.

glynhudson, 2019-10-24:
[image]
[image]

thedumbterminal, 2020-01-01:
I'm experiencing this when the device i'm using to display the graph gets disconnected from its wifi. I've created the pull request above to fix this.

thedumbterminal, 2020-01-01:
Here are the steps to recreate this problem:
1) Load up the solar app in your browser, get a graph showing your data.
2) turn off your wifi or unplug your ethernet
3) click on one of the date ranges such as "3h", "1d" etc (or wait for data to refresh).
4) You'll see the error appear.

thedumbterminal, 2020-01-06:
@glynhudson Could you take a look at the PR linked above, thanks.

thedumbterminal, 2020-01-06:
@emrysr hi any chance you could take a look at the above PR?

emrysr, 2020-01-08:
@thedumbterminal - I've tested it and all seems fine. not sure what the use case is when disconnected from the server? no new data will come through. But useful to clear the error. Thanks.

here's the error message I saw before testing the pull request:
[image]

thedumbterminal, 2020-01-09:
@emrysr not so much a use case but i'm running a raspberry pi with a touch screen on the wall which constantly shows the solar app via chromium in kiosk mode.

The problem occurs when the pi loses wifi connection (i think it is on the edge of the router's range). if the solar (or any) app tries to refresh its data when the pi has got disconnected temporarily then the above error message pops up, which blocks other javascript from running and someone may need to click through a number of popups (depending on the duration of the wifi outage) to get the solar app to refresh its data again.

Ideally I would have better wifi, but I don't think that a popup should appear on these issues either.

Thanks for making an amazing product!

closed 2020-01-22 by TrystanLea