emoncms/app #112: adding 'display as kW' in MyElectric2

emrysr, 2019-09-10:
I'vs noticed an issue on MySolarDivert, the exporting figure is very incorrect. The feed values has not changed between these two screenshots:

[image]

[image]

_Originally posted by @glynhudson in https://github.com/emoncms/app/pull/109#issuecomment-528113370_

emrysr, 2019-09-10:
@glynhudson - this should fix it, however could you test with your data and let me know if the problem persists? 

thanks

these are screenshots with my 'fake' data:

[in Watts]
[image]

[in Kilowatts]
[image]

glynhudson, 2019-09-11:
It looks good apart from the divert figure is showing decimal places. 

I think it would be good if the values showed one decimal place e.g 1.6kW. This will still take up less space than 1600W. Would you like me to make a new issue for this?

closed 2019-09-11 by glynhudson

emrysr, 2019-09-11:
No that's fine. I'll create a pull request to this issue and re open it when I start

PR #113 (2019-09-11): removed w_to_kw() function
corrected mysolardivert's import/export value

fix #112