emoncms/app #135: Invalid dimensions for plot Octopus Agile app on iphone

TrystanLea, 2020-02-26:
See: https://community.openenergymonitor.org/t/octopus-agile-emoncms-app/11767/89

whatsupskip, 2020-04-04:
I am seeing essentially the same error on Android when using Chrome to view the "Solar App" page. It happens if the phone is slightly rotated or rotated 90 degrees. This is for EmonCMS running locally on RPi on version low-write 10.2.1 & low-write 10.2.0. I don't believe the error was occurring on 10.1.13.

glynhudson, 2020-04-07:
I've also noticed this, the issue effects most apps including MyElectric and MySolar PV

[image]

https://photos.app.goo.gl/FEaDynSYivoVPKDU8

glynhudson, 2020-04-08:
Issue also effects the new version of emoncms.org apps 

[image]

glynhudson, 2020-04-08:
Issue has been fixed https://github.com/emoncms/app/commit/f3bbb297b65b5a84cabf4e32cfd9074ce7807e91

closed 2020-04-08 by glynhudson