emoncms/app #158: My Solar PV Battery - Undefined object error when SOC field disabled

jernaug, 2021-05-10:
My Solar PV Battery - Undefined object error when SOC field disabled
Reported by Neil Oakley

EmonCMS Error
Message: TypeError: undefined is not an object (evaluating ‘feeds[config.app.battery_soc.value].value’)
Route: app/view?name=My+Solar+Battery2
Line: 722
Column: 71

jernaug, 2021-05-10:
I’m able to reproduce the issue by disabling the SOC feed too. The issue seems to be due to
if (config.app.battery_soc.value)
evaluating to true when battery_soc has been disabled in the settings page. For me this then disabled access to the settings view when subsequently trying to access the app.

I managed to get access back to the settings view by modifying line 381 in mysolarpvbattery.php to set the optional property to false instead of true.
“battery_soc”:{“optional”:false, “type”:“feed”, “autoname”:“battery_soc”, “engine”:“5”, “description”:“Battery state of charge %”},

FredM67, 2021-06-11:
When one of the feed is disabled, then the graph is broken...
Tried with "Wind"...

TrystanLea, 2026-10-06:
Hello @FredM67 @jernaug, apologies for slow reply! My Solar Battery is now archived. My Electric Flow treats the SOC feed as optional and checks it before use.

closed 2026-10-06 by TrystanLea