emoncms/app #172: Incorrect (out of date?) Octopus Go rates

reedy, 2022-08-25:
Looking at the Octopus Agile app, the Go rates (for Yorkshire) seem rather out of date, using 5p/kWh and 16.67kW/h.

My fixes from March 2022 are:
[image]

Which are obviously quite a bit different, and therefore make it hard to compare to whether Agile makes sense...

glynhudson, 2022-08-26:
Daytime rates are 43.29p/kWh in North Wales :-( 

The issue is some people are still on the old tariff, it's probably easier to let people input their own rates. It would also be good to make the off-peak period configurable, I'm on Go Faster, so my off-peak period is at a different time.

reedy, 2022-08-26:
Yeah, I would concur allowing people to select their off peak time and enter their rates is probably the best way forward

reedy, 2022-08-27:
Looks like @TrystanLea beat us to it c2d5d971a8e4a74b76e8a18d97aca5bd73daa9d4

closed 2026-10-06 by TrystanLea