emoncms/app #182: MyHeatpump year view on energy page

glynhudson, 2023-04-20:
It would be very useful to have a 'Year' view option on the energy page. As an advanced function it would also be useful to have a date picker to view data in-between two dates: 

https://community.openenergymonitor.org/t/introducing-heatpumpmonitor-org-a-public-dashboard-of-heat-pump-performance/21885/211?u=glyn.hudson

MrTimbones, 2024-02-08:
YEAR was added by ca1c33b2bf39158056fea04a783b2f64c4025431, with DAY and 3 MONTHS added shortly after.

[image]

Issue can be closed.

TrystanLea, 2026-10-06:
Year, day and 3 month views were added in ca1c33b and later commits. A custom start and end window is also available from the time bar. Closing.

closed 2026-10-06 by TrystanLea